fboundy/ha_solis_modbus #2: Missing input?

finalbillybong, 2022-12-19:
Hey. Great work on this. Thank you!

The template below references 'input_number.solis_battery_capacity'

 ```
 - sensor:       
     name: "Solis Eco7 Charge Current"
     unique_id: "Solis Eco7 Charge Current"
     device_class: current
     state_class: measurement
     unit_of_measurement: A
     state: >-
       {{(
       ((states('input_number.solis_battery_capacity') | int * 1000) *
       ((states('input_number.eco7_target_soc') | int) - (states('sensor.solis_battery_soc') | int))
       ) 
       / 10
       / 
       (states('sensor.solis_eco7_time_remaining') | float)
       / (states('sensor.solis_battery_voltage_bms') | float) ) | int / 10
       }}
```

But I cannot see that in the inputs.yaml? My HA throws the error 'Template error: int got invalid input 'unknown' when rendering'.

fboundy, 2023-01-05:
Noted

fboundy, 2023-01-05:
Fixed in v1.1.2

closed 2023-01-05 by fboundy