fboundy/ha_solis_modbus #4: S2-WL-ST - DataLogger Problems

alienatedsec, 2023-01-27:
Thanks for the great work @fboundy 

Has anyone experienced issues with cloud disconnecting whenever you start using an integration - no matter if this one, I mean any integration that connects to port 502?

fboundy, 2023-01-27:
It certainly disconnects with this and I can’t see why it would be different with any other app using 502. I suspect it is something like a conflict of masters on the ModBus.

alienatedsec, 2023-01-29:
Ok. I am currently working on Slave to Multi-Master communication and will inform the progress.

fboundy, 2023-02-07:
I'm going to leave this as an Open issue as it is a genuine limitation which as far as I can see is hardware/firmware related. Let me know if you have any success/

alienatedsec, 2023-02-09:
After many attempts, I got a draft of the solution and waiting for some additional HW to be delivered today. Once it is built, I will report on the outcome.

alienatedsec, 2023-02-09:
@fboundy I have made the below set-up, which seems to be a combination of the Solis Cloud and the local Home Assistant integration.

[image]

I no longer use the `S2-WL-ST`, but it works with the above and will remain as my backup datalogger. Waveshare has what I need from the Modbus TCP communication to use integrations like yours or the SolaX Modbus, which also works with Solis Hybrid inverters. 

The `DLS-W` is enough for my reporting and to report back to Solis Cloud, especially since I made a lot of effort to integrate it with `SolisMon3` and `Grafana`.

If you have any questions, please let me know. I am currently testing how stable it is.

asos-timmills, 2023-02-20:
@alienatedsec Any details you can share on the hardware/configuration for the Waveshare Modbus TCP Server/Client please

alienatedsec, 2023-02-20:
@asos-timmills 

I am using the Waveshare RS485 to ETH - PoE version - the non-PoE version is described in option 1 of [this guideline](https://github.com/wills106/homeassistant-solax-modbus/wiki/Installation-Notes). There are no major differences between PoE and non-PoE devices, except for the convenience of powering those when you have multiple next in line.

The `TCP Server` on my diagram acts as a gateway for other devices (datalogger and HA integration), and it has a baud rate of 9600. The `Modbus gateway type` is `Auto query storage type`, and I enabled `multi-host`, so the datalogger queries will not create conflicts - at least that is my reasoning.

The `TCP Client` is pointing at the IP of the `TCP Server` with the same Serial config.

Both `TCP Server` and `TCP Client` have the `Modbus_TCP Protocol` as the `Transfer Protocol`

The Waveshare used for the batteries connection is the same device, but the `Transfer Protocol` is set to `None`. I use VirCom to extend its connection to COMX on my laptop, so the software can connect to batteries and monitor them. Looking into using `Modbus_TCP Protocol` and integrating with HA via MQTT or even setting the MQTT on the Waveshare, but that is not my priority at the moment.

I am trying different settings as had some teething issues, and I noticed that it works better when the integration (I am using [SolaX integration](https://github.com/wills106/homeassistant-solax-modbus) for my Solis inverter) is pulling data more often (e.g., every 5 seconds). Also, I am playing with delays and conflict settings, so as said earlier, I am still testing how stable it is with different configuration settings.

alienatedsec, 2023-03-07:
@fboundy happy for you to move this issue to discussions or simply close it.

I also made a repo based on my experience and will keep documenting at the following address https://github.com/alienatedsec/solis-ha-modbus-cloud

fboundy, 2023-03-13:
I think your solution may well be the only way to keep SolisCloud and Home Assistant running together. From what I can see as soon as a socket connects to the S2 on TCP port 502 SolisCloud is shut out so I have to assume that it is also using 502. The solution would be for the DataLogger to expose a secondary Modbus connection on another port which is presumably what the DLS-L does on port 8899.

closed 2023-03-13 by fboundy