fboundy/ha_solis_modbus #5: Modbus Unpack Errors

simon-talbot, 2023-02-06:
I am using the script against a Solis Hybrid invertor and most sensors are working, but I see the following sensors which are not updating correctly:

Solis Total Generation
Solis Timed Discharge 2 End Minute
Solis Temperature
Solis Grid Active Power

Which are all showing as unavailable (they are sometimes intermittent)

I am also seeing messages of the type:

2023-02-06 17:26:05.303 ERROR (MainThread) [homeassistant.components.modbus.base_platform] Received 20 bytes, unpack error unpack requires a buffer of 2 bytes
2023-02-06 17:26:05.305 ERROR (MainThread) [homeassistant.components.modbus.base_platform] Received 20 bytes, unpack error unpack requires a buffer of 2 bytes
2023-02-06 17:26:05.308 ERROR (MainThread) [homeassistant.components.modbus.base_platform] Received 20 bytes, unpack error unpack requires a buffer of 2 bytes

In the log for Modbus and getting one python error as follows:

fboundy, 2023-02-06:
What datalogger are you using?

simon-talbot, 2023-02-06:
I am using a DLS-L.

I have managed to find one of there errors which was causing the problem with “Solis Timed Discharge 2 End Minute”, there is a typo in solis.yaml, the address should be 431560, not 43160

    - name: Solis Timed Discharge 2 End Minute
      slave: 1
      address: 431560 <=====
      count: 1
      input_type: holding
      data_type: uint16
      lazy_error_count: 5
      scan_interval: 60
      scale: 1
      precision: 0
      state_class: measurement

But I am still left with three errors in the log:

2023-02-06 20:56:32.503 ERROR (MainThread) [homeassistant.components.modbus.base_platform] Received 4 bytes, unpack error unpack requires a buffer of 2 bytes
2023-02-06 20:56:32.506 ERROR (MainThread) [homeassistant.components.modbus.base_platform] Received 4 bytes, unpack error unpack requires a buffer of 2 bytes
2023-02-06 20:56:32.508 ERROR (MainThread) [homeassistant.components.modbus.base_platform] Received 4 bytes, unpack error unpack requires a buffer of 2 bytes

And I can’t find out an easy to diagnose which sensors these correspond to, I am guessing that it may be a sensor configured as an int16, that is actually an int32, but I can’t work out how easily to track down which ones the errors relate to.

Also, some sensors seem to be unreliable, flipping back and forth between a correct value and unavailable (such as Daily Consumption, temperature and grid frequency, but I thought I would try to get rid of the decode errors first, in case they are throwing things off?

Simon

fboundy, 2023-02-07:
Good catch on the address though it should be 43156 not 431560. Let me check the others that are causing problems

simon-talbot, 2023-02-07:
Thank you, you may want to double check on the correct address, it look to me like “Solis Timed Charge 2 End Minute” would be 43156 whereas the one with the error I picked up, “Solis Timed Discharge 2 End Minute”, should be 43160?

Simon

simon-talbot, 2023-02-07:
For your reference, I am also getting odd readings on some of the totals:

***@***.***

Simon

fboundy, 2023-02-07:
Your errors may be at least partly down to the DLS-L – I’m currently using an S2 and I’m not getting them. I do have a DLS-L so I will see what I get using that when I have a chance

fboundy, 2023-02-07:
OK - fixed the address for the timed discharge. I'd mis-read your original comment. It should be 43160 but it was in the YAML as 431560.

The `data type` for the other entities appear to be correct and are working OK with my S2 datalogger. 

The error says that you are receiving 20 bytes when 2 are expected. That suggests to me that `count` is set to 20 when expecting a `data type = uint32` but there are none set up that way in the YAML.. My guess is that it may be a firmware issue with either the DLS-L or your inverter but it is just a guess. What are the FW versions? I will check them against mine.

simon-talbot, 2023-02-07:
Interestingly, I have just reduced the scan_interval all of the sensors in solis.yaml that had an interval of 3600 down to 300 and all the errors seem to have gone away (for the moment). All sensors are now showing what look loke correct values and the messages have disappeared from the log, all a little odd!

I still have one recurring message remaining in the log, which is:

2023-02-07 10:49:01.332 ERROR (SyncWorker_6) [homeassistant.components.modbus.modbus] Pymodbus: PCInv1: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

That happens every 1 minute, so again a bit of a needle in a haystack, but a definite improvement!

Simon

Simon Talbot MEng(Hons) ACGI MIET
Managing Partner
Talbot Property Partners
T: 01206 982292
M: 07768 366670

The information contained in this e-mail and any attachments are private
and confidential and may be legally privileged.

It is intended for the named addressee(s) only. If you are not the
intended recipient(s), you must not read, copy or use the information
contained in any way. If you receive this email or any attachments in
error, please notify us immediately by e-mail and destroy any copy you
have of it.

We accept no responsibility for any loss or damages whatsoever arising
in any way from receipt or use of this e-mail or any attachments. This
e-mail is not intended to create legally binding commitments on our
behalf, nor do its comments reflect our corporate views or policies.

simon-talbot, 2023-02-07:
For completeness, that last message is not every minute, it is every minute for a few minutes, then a seven minute pause, then every minute again, so possible they are just an intermittent timeout, I will log it for longer and see if there is a pattern!

Simon

fboundy, 2023-02-07:
Yes - I've switched to my DLS-L and I'm now seeing the same thing.

simon-talbot, 2023-02-07:
Interesting, when you say an S2, do you mean an S2-WL-ST: WiFi logger with LAN connection (ie. The brand new one that they just started selling in January 2023 with the black body and the waterproof gland etc. for the Cat6 Cable)? The reason I ask is that I had one of these, but I could not get it to work at all and from what I read it seemed it was no longer directly accessible with MODBUS TCP/IP – It is of course possible that the S2-WL-ST I had was duff, as it kept going offline after about 5 minutes and needed to be power cycled? This was actually why I sought out someone who had old stock to buy a DLS-L !!

Simon

fboundy, 2023-02-07:
Yes - exactly one of those. I got it as a freebie from Solis. It works fine on TCP port 502 but I access SolisCloud when it is connected to HA.

simon-talbot, 2023-02-07:
I have turned debug up full on HA and thought the following might help identify the particular call that is failing:

2023-02-07 12:52:23.739 DEBUG (SyncWorker_1) [pymodbus.transaction] Running transaction 136
2023-02-07 12:52:23.739 DEBUG (SyncWorker_1) [pymodbus.transaction] SEND: 0x0 0x88 0x0 0x0 0x0 0x6 0x1 0x3 0xa8 0x3 0x0 0x1
2023-02-07 12:52:23.739 DEBUG (SyncWorker_1) [pymodbus.client.base] New Transaction state "SENDING"
2023-02-07 12:52:23.740 DEBUG (SyncWorker_1) [pymodbus.transaction] Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-02-07 12:52:24.038 DEBUG (SyncWorker_1) [pymodbus.transaction] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-02-07 12:52:24.038 DEBUG (SyncWorker_1) [pymodbus.transaction] RECV: 0x0 0x88 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x14
2023-02-07 12:52:24.038 DEBUG (SyncWorker_1) [pymodbus.framer.socket_framer] Processing: 0x0 0x88 0x0 0x0 0x0 0x5 0x1 0x3 0x2 0x0 0x14
2023-02-07 12:52:24.038 DEBUG (SyncWorker_1) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse': 3]
2023-02-07 12:52:24.039 DEBUG (SyncWorker_1) [pymodbus.transaction] Adding transaction 136
2023-02-07 12:52:24.039 DEBUG (SyncWorker_1) [pymodbus.transaction] Getting transaction 136
2023-02-07 12:52:24.039 DEBUG (SyncWorker_1) [pymodbus.transaction] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

2023-02-07 12:53:03.064 DEBUG (SyncWorker_0) [pymodbus.transaction] Running transaction 137
2023-02-07 12:53:03.064 DEBUG (SyncWorker_0) [pymodbus.transaction] SEND: 0x0 0x89 0x0 0x0 0x0 0x6 0x1 0x4 0x81 0x45 0x0 0x1
2023-02-07 12:53:03.065 DEBUG (SyncWorker_0) [pymodbus.client.base] New Transaction state "SENDING"
2023-02-07 12:53:03.065 DEBUG (SyncWorker_0) [pymodbus.transaction] Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-02-07 12:53:06.069 DEBUG (SyncWorker_0) [pymodbus.transaction] Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received))
2023-02-07 12:53:06.069 DEBUG (SyncWorker_0) [pymodbus.framer.socket_framer] Processing:
2023-02-07 12:53:06.069 DEBUG (SyncWorker_0) [pymodbus.transaction] Getting transaction 137
2023-02-07 12:53:06.069 DEBUG (SyncWorker_0) [pymodbus.transaction] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2023-02-07 12:53:06.069 ERROR (SyncWorker_0) [homeassistant.components.modbus.modbus] Pymodbus: PCInv1: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2023-02-07 12:53:06.069 DEBUG (SyncWorker_0) [pymodbus.transaction] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2023-02-07 12:53:06.069 ERROR (SyncWorker_0) [homeassistant.components.modbus.modbus] Pymodbus: PCInv1: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

2023-02-07 12:53:06.078 DEBUG (SyncWorker_2) [pymodbus.client.tcp] Connection to Modbus server established. Socket ('10.51.2.116', 37140)
2023-02-07 12:53:06.078 DEBUG (SyncWorker_2) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2023-02-07 12:53:06.078 DEBUG (SyncWorker_2) [pymodbus.client.tcp] Connection to Modbus server established. Socket ('10.51.2.116', 37140)
2023-02-07 12:53:06.078 DEBUG (SyncWorker_2) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2023-02-07 12:53:06.078 DEBUG (SyncWorker_2) [pymodbus.transaction] Running transaction 138
2023-02-07 12:53:06.079 DEBUG (SyncWorker_2) [pymodbus.transaction] SEND: 0x0 0x8a 0x0 0x0 0x0 0x6 0x1 0x4 0x82 0x2 0x0 0x1
2023-02-07 12:53:06.079 DEBUG (SyncWorker_2) [pymodbus.client.base] New Transaction state "SENDING"
2023-02-07 12:53:06.080 DEBUG (SyncWorker_2) [pymodbus.transaction] Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-02-07 12:53:09.083 DEBUG (SyncWorker_2) [pymodbus.transaction] Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-02-07 12:53:09.083 DEBUG (SyncWorker_2) [pymodbus.transaction] RECV: 0x0 0x8a 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x13 0x96
2023-02-07 12:53:09.084 DEBUG (SyncWorker_2) [pymodbus.transaction] Got response!!!
2023-02-07 12:53:09.084 DEBUG (SyncWorker_2) [pymodbus.framer.socket_framer] Processing: 0x0 0x8a 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x13 0x96
2023-02-07 12:53:09.084 DEBUG (SyncWorker_2) [pymodbus.factory] Factory Response[ReadInputRegistersResponse': 4]
2023-02-07 12:53:09.084 DEBUG (SyncWorker_2) [pymodbus.transaction] Adding transaction 138
2023-02-07 12:53:09.085 DEBUG (SyncWorker_2) [pymodbus.transaction] Getting transaction 138
2023-02-07 12:53:09.085 DEBUG (SyncWorker_2) [pymodbus.transaction] Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

2023-02-07 12:53:09.087 DEBUG (SyncWorker_4) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2023-02-07 12:53:09.088 DEBUG (SyncWorker_4) [pymodbus.transaction] Running transaction 139

It seems to be the call that is made in transaction 137 above that repeatedly fails, with the packet being sent being SEND: 0x0 0x89 0x0 0x0 0x0 0x6 0x1 0x4 0x81 0x45 0x0 0x1, which to me looks like register 0x8145 or 33,093 which is Inverter temperature that is causing the issue. Very odd because the inverter temp is reading in HA, so must be intermittent, all a bit odd!

Simon

simon-talbot, 2023-02-07:
Ahh, that will be the issue then, I was using port 8899 when I tried S2-WL-ST as I hadn’t found your great project! That said, it shouldn’t have gone offline after 5 minutes, so suspect it was duff anyway, will have to get another one and give it a try if it is a bit more robust!

Simon

alienatedsec, 2023-02-14:
> Ahh, that will be the issue then, I was using port 8899 when I tried S2-WL-ST as I hadn’t found your great project! That said, it shouldn’t have gone offline after 5 minutes, so suspect it was duff anyway, will have to get another one and give it a try if it is a bit more robust!

There is another limitation with those new Dataloggers `S2-WL-ST`. If you want to use them with HA, connecting to port 502 will disconnect Solis Cloud and stop reporting. See #4

simon-talbot, 2023-02-14:
Ahh, thank you for that, will stick with the DLS-L then, which for reference now seems to be working OK with the reduced timers. Looking at the logs, it is still erroring/timing out intermittently, but with the shorter refresh timers I introduced, HA is still getting updated frequently enough to function well. It also continues to function with Solis cloud in parallel.

There is another limitation with those new Dataloggers S2-WL-ST. If you want to use them with HA, connecting to port 502 will disconnect Solis Cloud and stop reporting. See #4<https://github.com/fboundy/ha_solis_modbus/issues/4>

fboundy, 2023-03-13:
I will close this then - it does seem to be a bug with the DLS-L

closed 2023-03-13 by fboundy

simon-talbot, 2023-06-16:
Hiya, just a quick question that I hope you may have an easy answer to, I have finally got myself a S2-WL-ST to test on one of my inverters but am having difficulty working out what the correct username/password combination is to log into the UI. The docs say Admin/123456789 which doesn’t work, but I seem to remember the last time I briefly played with one of these that either the username or the password was the mac address or serial number, but I can’t for the life of me work out the correct combination, do you happen to remember?

Best regards,

Simon

Simon Talbot MEng(Hons) ACGI MIET
Managing Partner
Talbot Property Partners
T: 01206 982292
M: 07768 366670

The information contained in this e-mail and any attachments are private
and confidential and may be legally privileged.

It is intended for the named addressee(s) only. If you are not the
intended recipient(s), you must not read, copy or use the information
contained in any way. If you receive this email or any attachments in
error, please notify us immediately by e-mail and destroy any copy you
have of it.

We accept no responsibility for any loss or damages whatsoever arising
in any way from receipt or use of this e-mail or any attachments. This
e-mail is not intended to create legally binding commitments on our
behalf, nor do its comments reflect our corporate views or policies.

Closed #5<https://github.com/fboundy/ha_solis_modbus/issues/5> as completed.

alienatedsec, 2023-06-16:
> The docs say Admin/123456789 which doesn’t work

The docs say lowercase letters

simon-talbot, 2023-06-16:
Oh god, I have tried absolutely every combination of Serial Numbers, MAC Address, Admin 123456789, but of course, the one thing I didn’t try was a lower case a!!!

The docs I received with the unit very clearly stated an upper case Admin, but if you look carefully at the screenshot, it shows a lower case, typical!!!:

***@***.***

Thank you so much for preventing me wasting hours.

Simon