fboundy/ha_solis_modbus #7: Where is sensor.solis_eco7_time_remaining?

SGDAT, 2023-04-11:
I ran into this error related to automations before I realised I couldn't find a reference to `sensor.solis_eco7_time_remaining`.

```
Error while processing template: Template<template=({{( ((states('input_number.solis_battery_capacity') | int * 1000) * ((states('input_number.eco7_target_soc') | int) - (states('sensor.solis_battery_soc') | int)) ) / 10 / (states('sensor.solis_eco7_time_remaining') | float) / (states('sensor.solis_battery_voltage_bms') | float) ) | int / 10 }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1888, in forgiving_float_filter
    return float(value)
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 539, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2130, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1891, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1589, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{( ((states('input_number.solis_battery_capacity') | int * 1000) * ((states('input_number.eco7_target_soc') | int) - (states('sensor.solis_battery_soc') | int)) )  / 10 /  (states('sensor.solis_eco7_time_remaining') | float) / (states('sensor.solis_battery_voltage_bms') | float) ) | int / 10 }}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 660, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 541, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{( ((states('input_number.solis_battery_capacity') | int * 1000) * ((states('input_number.eco7_target_soc') | int) - (states('sensor.solis_battery_soc') | int)) )  / 10 /  (states('sensor.solis_eco7_time_remaining') | float) / (states('sensor.solis_battery_voltage_bms') | float) ) | int / 10 }}' but no default was specified

```

I am admittedly new at home assistant programming although I am quite experienced with software dev.

finalbillybong, 2023-04-14:
The sensor you’re looking for is called sensor.solis_eco7_hours_remaining I think it is mislabelled in the dashboard yaml

fboundy, 2023-06-21:
Good catch. I've changed the template so that the name and ID are consistent.