fboundy/ha_solis_modbus #8: Support for controlling battery to grid discharge

finalbillybong, 2023-04-14:
Would you consider adding support for controlling the discharge rates of the battery for those of us that are on Octopus Flux?

finalbillybong, 2023-04-15:
https://64.media.tumblr.com/3a06bb4919f3ac8536d1817270670264/tumblr_nwhax4ceuy1ur9h4do1_500.gif

I have added support for this.

It's built around the same principle that you have used for the Eco 7 Charging. I have been using it for a couple of days and it works quiet well. 

I have also created automations, sensors and inputs to work out the ideal overnight charge percentage (if any) based on household usage and the day ahead predicted solar output. Happy to create a pull request if you would like.

fboundy, 2023-04-15:
Thanks - need to catch up on this!

fboundy, 2023-04-17:
Fire ahead with the PR - I've done something similar but mine is rather complex compared to the approach taken in this repo

finalbillybong, 2023-04-17:
Sure thing. I have not commited the changes for Flux just yet as I'm still testing but all the bits for overnight charging are in there.

solarjones, 2023-05-12:
I'm quite new to HA and struggling to work out where to start to install either of your work. I'm on Octopus Eco7, thinking about switching to Flux and only have a noddy automation looking at SolCast next day solar forecast to then set charging times based on Good, Fair, Bad forecasts. I'd love to setup something with more intelligence. Any help or additional instructions/pictures etc would be appreciated.

closed 2023-10-21 by fboundy