fboundy/ha_solis_modbus #11: MODBUS TCP/IP Issue with S2-WL-ST data loggers and older firmware

simon-talbot, 2023-06-27:
Whilst this is a self resolved issue, I thought it worthwhile sharing it, in the hope it will help others save a lot of wasted time!

I have been working with Solis on a problem with MODBUS TCP/IP not functioning on some of our S2-WL-ST data loggers with older firmware (firmware version10010106 in my case). Once the firmware of the S2-WL-ST was upgraded to one of the current versions (Version 10010117 in our case) MODBUS TCP/IP functionality worked correctly and reliably, so there is definitely a problem with the older firmware. For reference, the very latest firmware (10010119) would not install on the sticks with the old (10010106) firmware and crashed the stick (Fixed Red & Amber LED) whenever the upgrade was attempted by Solis support, they then tried the 10010117 firmware and it installed correctly and MODBUS TCP/IP functioned correctly, so all in all rather painful!

simon-talbot, 2023-06-28:
As a subscript to this, when the 10010106 firmware was first upgraded to 10010117, it was then possible to upgrade it to 10010119 subsequently without crashing the S2-WL-ST.

fboundy, 2023-06-28:
Thanks for the update. As an aside - are you able to access SolisCloud at the same time as HA with that version or is it still either/or?

simon-talbot, 2023-06-28:
Unfortunately not, the behaviour is the same in that respect and it seems deliberate and intended. As soon as you have accessed the stick via Modbus, it stops attempting to communicating with Solis Cloud. I even tried only accessing every 30 minutes with Modbus, but it still did not resume communication with Solis. As far as I can tell, once you have accessed it with Modbus, it will not communicate with Solis cloud unless you restart it. Additionally I noticed that if you briefly press the reset button on the stick, which would usually initiate a data upload to the cloud, if you have accessed with Modbus, the yellow light just blinks slowly, which I assume indicates an error condition and no upload happens, which is why I say the action of either/or is intended coding, rather than a side effect.