fboundy/ha_solis_modbus #14: Modbus failing after upgrade to HA 2023.9

Paulholloway, 2023-09-09:
I've [raised the issue elsewhere as I believe that HA 2023.9 is where the issue really lies](https://github.com/home-assistant/core/issues/99784#issuecomment-1712455965), but have been asked to point the question back to yourself.

Whilst running:

> Home Assistant 2023.9.1
> Supervisor 2023.08.3
> Operating System 10.5
> Frontend 20230908.0 - latest
> 
> I'm repeatedly finding that modbus trips up:
> 
> 2023-09-09 08:51:30.434 ERROR (SyncWorker_1) [homeassistant.components.modbus.modbus] Pymodbus: solis: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
> 2023-09-09 08:59:51.205 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus solis communication closed
> 2023-09-09 09:00:07.420 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus solis communication closed
> 2023-09-09 09:00:07.715 ERROR (SyncWorker_18) [pymodbus.logging] Connection to (IP ADDRESS, 502) failed: [Errno 111] Connection refused
> 2023-09-09 09:01:22.378 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus solis communication closed
> 

I've increased `lazy error count `to 10 on each of the sensors to see if it lasts a little longer with a mixed bag of results.  It first bought me five minutes but has been stable for ~ 5 hours since.  In each instance, the problem is resolved with a config reload.  

No issues with the dongle, which can be accessed via its IP address, restarted etc.  

Any help would be appreciated.

fboundy, 2023-09-09:
As I mentioned on the other thread, my repo isn’t actually a custom integration- just a set of a scripts and configs that call the standard HA ModBus service so the issue must either be with that or with the Solis hardware/firmware.

I’m away at present so I will investigate when I return. Have you tried polling the dongle using the Python ModBus script in my repo?

Paulholloway, 2023-09-09:
I'm not sure how this repo was misinterpreted as an integration (not my words).  The simplicity is why I was certain that the issue was not with your code and so decided to start asking questions elsewhere to begin with.  

I've not tried polling (I see the additional files but, beyond adding to my config, I'm not sure what else is needed or if they get to work once reloaded - sorry).  

I've had greater luck today and managed most of the day before the sensors all became unavailable.  Same error:

```
2023-09-09 19:01:26.383 ERROR (SyncWorker_20) [homeassistant.components.modbus.modbus] Pymodbus: solis: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

2023-09-09 19:06:32.003 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus solis communication closed
```

The increased the lazy error count might be what's helping?  

I've got a quick and dirty automation that triggers ` homeassistant.reload_all`, so that the sensors promptly come back to life as soon as they appear unavailable.    

It will be interesting to see if there actually is a [fix in the pipeline](https://github.com/home-assistant/core/pull/99988)... 

Thanks again.

fboundy, 2023-09-13:
This is working for me with 2023.9.2. Can you confirm if this is the same for you and I will close

Paulholloway, 2023-09-13:
Yes, this has been solid for most of the day now so appears to be fixed!  Thanks for your input!