fboundy/ha_solis_modbus #17: Home Assistant version 23.20.0 Modbus Keyword "Count" depreciation.

nilremgithub, 2023-10-05:
It seems the keyword count no longer supported for the data types  int16 or uint16 or int32 or uint32 in the solis.yaml file. Further info here: https://github.com/home-assistant/core/issues/101421

mrstevefay, 2023-10-16:
Is their a fix for this we can implement?

lorrainem77, 2023-10-16:
In the modbus yaml file, find all "count:" lines and either put a # in front of it (e.g. #count: 1) or just remove the line

fboundy, 2023-10-16:
Yes, that’s correct. Will update the repo when I get a minute

Paulholloway, 2023-10-21:
I'm genuinely putting off ever doing another core update... This is the second time in a row that they've broken modbus!

fboundy, 2023-10-21:
Closed with release 2.0.0

closed 2023-10-21 by fboundy

joeczucha, 2023-10-25:
> Closed with release 2.0.0

Just a heads up that where you've done a find/replace on `count:` you've also caught `lazy_count:` so now your solis.yaml is reading: `lazy_error_# count: 5` and throwing errors.

Otherwise, great project :)

fboundy, 2023-10-25:
Doh! That’ll teach me. Will get it fixed asap.

fboundy, 2023-10-26:
Now fixed