fboundy/ha_solis_modbus #18: Connection refused

remoteboy, 2023-10-30:
Hi, 
I have a Solis inverter - RHI-5k-48ES - with a wired datalogger stick - S2-WL-ST. I’m trying to get it to talk to my home assistant install. I added the `solis.yaml `file to my HA install and added the IP address to the `secrets.yaml` file and added `modbus: !include solis.yaml` to my `configuration.yaml` file

I’ve restarted my HA and my log files are filling up with this error
`2023-10-30 11:59:43.906 ERROR (SyncWorker_4) [pymodbus.logging] Connection to (192.168.2.232, 502) failed: [Errno 111] Connection refused`. I’ve confirmed that the IP address listed above is the correct IP - I can access to via the web admin. I've removed the datalogger from my SolisCloud account to rule out any potential conflicts and I've tried both port 502 and port 8899. Not sure if I've missed a step somewhere along the way. Do I need to add credentials somewhere?

Thanks.

fboundy, 2023-10-30:
Hi - that all sounds right. What version of HomeAssistant Core are you running? Are you able to run pymodbus_test from Python?

remoteboy, 2023-10-30:
Yep. I'm getting connection refused when I run it. I've tried it from a few machines, not just the ha install. Do i need to make any changes to the Solis inverter itself to support this?

fboundy, 2023-10-30:
No you shouldn’t - ModBus just talks to the Datalogger, not the inverter. What firmware version do you have on the DLS?

remoteboy, 2023-10-30:
10010106

remoteboy, 2023-10-30:
Just ran `nmap` on the datalogger and the only open port i can see is port 80. Checked 502 and 8899 and they both return as closed. Not sure if `nmap` is designed to pick up on modbus ports though.

fboundy, 2023-10-30:
OK - I'm not actually using my S2 at present so I'll need to plug it in to check what version mine is and whether it's exposing any ports to nmap.

fboundy, 2023-10-30:
OK - my F/W is 1001f109 which sound like it might be newer than yours. 

nmap only shows port 80, same as yours.

remoteboy, 2023-10-30:
thanks for looking into this. i reckon first port of call is to try to get the firmware upgraded. Solis don't seem to offer new firmware downloads for the sticks so going to drop them an email.

fboundy, 2023-10-30:
I have a feeling someone else had a similar issue. Are you on the Solis Facebook group? Worth asking there maybe?

remoteboy, 2023-10-30:
Thanks. I'll take a look.

remoteboy, 2023-10-31:
Ok. Turns out it was the firmware on the stick. I contacted Solis who upgraded the stick to 10010119 and everything started working. Thanks for all the help.

closed 2023-10-31 by remoteboy

fboundy, 2023-10-31:
Glad you got it sorted

JonaHalffman, 2023-11-20:
Same issue over here. How do I update the firmware of my S2-WL-ST?

Pho3niX90, 2023-11-29:
> Same issue over here. How do I update the firmware of my S2-WL-ST?

You need to get in touch with support, they do it remotely.

finalbillybong, 2023-12-02:
> > Same issue over here. How do I update the firmware of my S2-WL-ST?
> 
> You need to get in touch with support, they do it remotely.
> 
> I am running firmware 10010109, and also have intermittent issues, especially when writing. Will do upgrade today and see if it resolves the issue.

Upgrading to 10010119 will sort you out.

Pho3niX90, 2023-12-02:
> Upgrading to 10010119 will sort you out.

It unfortunately did not. However, I wrote an integration for solis, which combined multiple registers into single calls, this in fact resolved the issue completely. I am able to write to the solis with immediate response, as well as poll all data every 5 seconds without issues. For those intrested -> https://github.com/Pho3niX90/solis_modbus/