fboundy/ha_solis_modbus #22: lazy_error_count key

nilremgithub, 2024-01-09:
It seems "lazy_error_count" will be depreciated. I received the warning below today.

Please remove the lazy_error_count key from the modbus entry in your configuration.yaml file and restart Home Assistant to fix this issue. All errors will be reported, as lazy_error_count is accepted but ignored

Paulholloway, 2024-01-13:
Is there an alternative to lazy error count or is it as simple as commenting it out?

nilremgithub, 2024-01-13:
I'm not sure. Since commenting out the line and restarting HA I don't get the warning anymore and the Modbus communication seems to be working fine.

fboundy, 2024-01-13:
There’s no longer any mention of it on the HA integration page so I think it just needs commenting out. Feel free to submit a PR or I will fix it next week when I have time

fboundy, 2024-01-31:
Removed in 2.1.0