fboundy/ha_solis_modbus #23: Solis Daily Import/Export no longer appear to reset

Paulholloway, 2024-04-05:
Having just updated to Core 2024.4.0 yesterday, I've just this morning noticed that my Solis Daily Import/Export entities have continued to increment beyond midnight.  

[image]

I'm not sure if this is a quirk, coincidence, or if I've just caught HA at the wrong time, but historical data showed a clear reset to zero at midnight.  Thoughts welcome!

fboundy, 2024-04-05:
These entities come directly from the inverter so I suspect it’s a Solis issue rather than an HA issue. I’m not seeing this with either Solax or HA core integrations bi TV I haven’t upgraded yet (always wait until at least x.x.1 to flush out bugs…)

Paulholloway, 2024-04-05:
Spot on and super quick response.  A quick look at the value of [solis day RW](url) reveals it was seemingly upset by the clocks changing.  Forcing an update to register 43002 fixed this, and my daily values have reset to zero.  

[image]
[image]

closed 2024-04-05 by Paulholloway