fboundy/ha_solis_modbus #24: Modbus integration not working reliably with HA Core 2024.4.1

nilremgithub, 2024-04-06:
The modbus integration has stopped working with the latest HA update. The log shows:

Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:383
integration: Modbus ([documentation](https://www.home-assistant.io/integrations/modbus), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+modbus%22))
First occurred: 11:51:50 (1 occurrences)
Last logged: 11:51:50

modbus solis communication closed

Paulholloway, 2024-04-06:
No issues observed here (and mine normally breaks quickly).  
Do you have the latest version?

nilremgithub, 2024-04-06:
Thanks for letting me know.  I will check the code I have added.

nilremgithub, 2024-04-07:
It seems other people are experiencing the same issue. It seems to be related to the 2024.4.1 upgrade.
See: https://github.com/home-assistant/core/issues/115040

fboundy, 2024-04-08:
A number of issues are now documented in https://github.com/home-assistant/core/issues/115236.

I'm not sure which, if any, relate to issues with the Solis inverter. If you are able to debug log the issue and submit it then that would be helpful but the guy who looks after this on HA is somewhat lacking in social skills so be warned!

spetrie13, 2024-04-08:
> A number of issues are now documented in home-assistant/core#115236

3.13.2

Using Solax for inverter communication.

Noted in logs also was getting 107A issues on writing to Inverter causing
modbus issues.
Changed inverter config in pv_opt from 5500W to 5000W so hopefully that
will limit charge to 100A limit of inverter.

Richardvdw01, 2024-05-13:
My modbus problems still persist after Bump pymodbus v3.6.7 #115279
Looking at the logs, it seems that the modbus connection keeps getting lost. This makes several entities unavailable at the frontend.
When rolling back to 2024.3.3 everything works flawless.
Here is the latest log when upgrading to 2024.5.3:

[Pymodbus fail.docx](https://github.com/fboundy/ha_solis_modbus/files/15299063/Pymodbus.fail.docx)

nilremgithub, 2024-05-14:
I am not experiencing any issues. However I added the following to the top of the solis.yaml file under the port setting to address the original concern and haven't removed it.

 # delay: 5
  message_wait_milliseconds: 100
  timeout: 5

Richardvdw01, 2024-05-16:
Thank you @ nilremgithub, Adding the delay resolved the issue. It looks like the new Pymodbus is too fast for the serial connection.

fboundy, 2024-05-16:
Please feel free the update the GitHub repo with a Pull Request incorporating this