fboundy/ha_solis_modbus #25: Probably a modbus issue, brought about by Core 2024.10

Paulholloway, 2024-10-13:
A compelling case not to upgrade core again, given the amount of times someone messes with python. I expect a fix will appear in due course but otherwise is there a tweak to the solis yaml possible to make it better resist whatever the problem is?

Since CORE 2024.10 (.1, .2) I now find that all of my solis sensors become unavailable on random occasions and the logs show no response after a number of retries. One goes and the rest follow in sequence. This normally resolves itself after a few minutes but can drop again not long after. It’s not clear to me whenever reloading modbus/restarting has a better effect than just waiting for time to pass. 

Log entries: 

```
Logger: pymodbus.logging
Source: runner.py:189
First occurred: 12 October 2024 at 22:44:41 (83 occurrences)
Last logged: 06:43:15

Failed to connect [Errno 111] Connect call failed ('192.168.1.xxx, 502)
Failed to connect
Unable to decode response Modbus Error: Unknown response 0
Failed to connect [Errno 113] Connect call failed ('192.168.1.xxx', 502)

Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:331
integration: Modbus ([documentation](https://www.home-assistant.io/integrations/modbus), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+modbus%22))
First occurred: 12 October 2024 at 22:44:40 (41 occurrences)
Last logged: 06:42:44

Pymodbus: solis: Error: device: 1 address: 43167 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Pymodbus: solis: Error: device: 1 address: 33079 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Pymodbus: solis: Error: device: 1 address: 43155 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Pymodbus: solis: Error: device: 1 address: 43169 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Pymodbus: solis: Error: device: 1 address: 33282 -> Modbus Error: [Input/Output] ERROR: No response received after 3 retries

Logger: homeassistant.components.modbus.entity
Source: components/modbus/entity.py:241
integration: Modbus ([documentation](https://www.home-assistant.io/integrations/modbus), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+modbus%22))
First occurred: 00:16:12 (4 occurrences)
Last logged: 05:58:23

Received 2 bytes, unpack error unpack requires a buffer of 4 bytes
Received 4 bytes, unpack error unpack requires a buffer of 2 bytes

```

Paulholloway, 2024-10-13:
Though I've now added in the delay/timeout from [Issue 24] to solis.yaml (https://github.com/fboundy/ha_solis_modbus/issues/24#issuecomment-2109545038)

```
  delay: 5
  message_wait_milliseconds: 100
  timeout: 5
```
I'll report back if that fixes things (and if CORE is just not being patient is the issue).

Paulholloway, 2024-11-20:
Seemingly stabilised now having increased the timeout (though I'm on HA Core 2024.10.3 and reluctant to make another update).  Whether this is all circumstantial... who knows!  

```
  delay: 10
  message_wait_milliseconds: 100
  timeout: 15
```

fboundy, 2024-11-20:
Be aware that the 4B HMI update to the inverter changes everything!

I will be working through a lot of updates to 3 of 4 integrations as a result…

Paulholloway, 2024-11-21:
To confirm - are these just the latest physical inverters or is this a software update to the existing stock?

Paulholloway, 2024-11-21:
Good job the modbus effectively closes the connection to SolisCloud etc. Hopefully you’ve a plan to overcome this?Sent from my iPhone

It’s an update that was pushed to my 2 year old inverter this week by Solis while they were fixing something else

fboundy, 2024-11-21:
Working on it

Paulholloway, 2024-11-21:
Star!Where’s the ‘buy a coffee’ for you link? Sent from my iPhone

fboundy, 2024-11-21:
https://ko-fi.com/fboundy